pascallanger/DIY-Multiprotocol-TX-Module #1081: Create new Protronik protocol

pierrotm777, 2025-03-03:
Hello, I have received one Ptr6a handset and I can send it to you for build a Protronik protocol. 
This little handset is nice because it can be used in mode 1 or mode 2 by switch. 
I should like use it in my club as trainer wireless for teach beginners. 
Thanks

pascallanger, 2025-03-03:
So there are 2 requests:
- Reverse the protocol, which I also need a RX for and not only the TX. I don't even know which RF component they use and if it's feasible...
- Implement a RX/Trainer protocol

That's a lot of work which will take weeks/months just to set expectations...

pascallanger, 2025-03-03:
Contact me on pascal _ langer @ yahoo . fr for the shipping address.

pascallanger, 2025-03-09:
I've received the package but no time to open it yet.

pierrotm777, 2025-05-14:
Hello Pascal,
Do you have some news about Protronik protocol 😊?