pascallanger/DIY-Multiprotocol-TX-Module #1082: Support for HotRC / Rlaarlo receivers

pascallanger, 2025-03-04:
### Discussed in https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/discussions/1072

<div type='discussions-op-text'>

<sup>Originally posted by **c0ldtech** February 11, 2025</sup>
It would be nice to have support for the HotRC receivers (which are also being used by Rlaarlo it seems). I'd be more than willing to order a receiver and have it sent to someone if it helps.</div>

pascallanger, 2025-03-04:
@c0ldtech I'll update my findings here

prikrylm, 2025-03-19:
I am willing to provide funds for the purchase of a test Tx-Rx. :) PayPal?

neutron-modulator, 2025-03-31:
> I am willing to provide funds for the purchase of a test Tx-Rx. :) PayPal?
Hi Guys 
I have a selection of HoRc Tx and Rx, I'm happy to test, wreck etc. I have pulled apart a DS-4A and stared a bit of reverse engineering. (I might need some guidance capturing signals off air with my SDR or JTAG dump of the firmware.)
Cheers Rod

neutron-modulator, 2025-05-16:
Hi Pascal 
I just received this from AliExpress: 
 ST-Link V2 https://a.aliexpress.com/_msNDlOB
Is it going to work to capture the data you require.

Cheers Rod

pascallanger, 2025-05-17:
No it's not the right device.... You need a cheap ($10) Saleae 8 channels compatible analyzer for that purpose.

neutron-modulator, 2025-05-17:
Cool I'll get that sorted.

baronKanon, 2025-07-29:
Is there any progress on this?

motorello, 2025-11-13:
Ok, it turned out that the RF transceiver is a LT8910, which is why the data payload is always 2 bytes. I'm pasting below the full initialization sequence for completeness. The configuration per se is not that bad, no scramble, no whitening, no FEC, easy hopping sequence. BUT: the data rate is 125Kbps, which is not supported by NRF24L01. This is bad since we cannot use the wonderful emulation layer that Pascal already developed in 2016, if I recall correctly.
I don't know, if someone managed to use some undocumented register in NRF24 to get 125Kbps data rate... Otherwise, I fear it is needed to write a similar emulation layer on top of CC2500. I will study the datasheet in spare time.

This is the full initialization SPI dump, together with some my notes:

| DELTA TIME us | SIGNAL | REG | R/W | MOSI REG | PKT-0 | PKT-1 | REG/REPL | BITS-P-0 | BITS-P-1 | NOTES |
| ---: | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- |
| 0 | MOSI | 00 | W | 00 | 6F | E0 | 00h (0d) | 01101111 | 11100000 | Recommended on datasheet |
| 0 | MISO | 00 |  | 00 | 00 | 01 | 00000000 | 00000000 | 00000001 |  |
| 709,773 | MOSI | 02 | W | 02 | 66 | 17 | 02h (2d) | 01100110 | 00010111 | Recommended on datasheet |
| 709,773 | MISO | 01 |  | 02 | 01 | 01 | 00000001 | 00000001 | 00000001 |  |
| 108 | MOSI | 04 | W | 04 | 9C | C9 | 04h (4d) | 10011100 | 11001001 | Recommended on datasheet |
| 108 | MISO | 01 |  | 04 | 80 | 01 | 00000001 | 10000000 | 00000001 |  |
| 106 | MOSI | 05 | W | 05 | 66 | 37 | 05h (5d) | 01100110 | 00110111 | Recommended on datasheet |
| 106 | MISO | 01 |  | 05 | 01 | 81 | 00000001 | 00000001 | 10000001 |  |
| 105 | MOSI | 07 | W | 07 | 00 | 00 | 07h (7d) | 00000000 | 00000000 | TX_EN: 0 RX_EN: 0 RF_CH: 0d / 00h |
| 105 | MISO | 01 |  | 07 | 01 | 01 | 00000001 | 00000001 | 00000001 |  |
| 106 | MOSI | 08 | W | 08 | 6C | 90 | 08h (8d) | 01101100 | 10010000 | Recommended on datasheet |
| 106 | MISO | 01 |  | 08 | 01 | 01 | 00000001 | 00000001 | 00000001 |  |
| 107 | MOSI | 09 | W | 09 | 1F | C0 | 09h (9d) | 00011111 | 11000000 | PA_PWCTR: 0001 (1d); PA_GN: 1111 (15d) |
| 107 | MISO | 01 |  | 09 | 81 | 81 | 00000001 | 10000001 | 10000001 |  |
| 108 | MOSI | 0B | W | 0B | 00 | 08 | 0Bh (11d) | 00000000 | 00001000 | Recommended on datasheet |
| 108 | MISO | 01 |  | 0B | 01 | 01 | 00000001 | 00000001 | 00000001 |  |
| 107 | MOSI | 0D | W | 0D | 48 | BD | 0Dh (13d) | 01001000 | 10111101 | Recommended on datasheet |
| 107 | MISO | 81 |  | 0D | 01 | 01 | 10000001 | 00000001 | 00000001 |  |
| 113 | MOSI | 16 | W | 16 | 00 | FF | 16h (22d) | 00000000 | 11111111 | Recommended on datasheet |
| 113 | MISO | 01 |  | 16 | 01 | 01 | 00000001 | 00000001 | 00000001 |  |
| 107 | MOSI | 17 | W | 17 | 80 | 05 | 17h (23d) | 10000000 | 00000101 | TxRx_VCO_CAL_EN: 1 |
| 107 | MISO | 00 |  | 17 | 00 | 01 | 00000000 | 00000000 | 00000001 |  |
| 108 | MOSI | 18 | W | 18 | 00 | 67 | 18h (24d) | 00000000 | 01100111 | Recommended on datasheet |
| 108 | MISO | 01 |  | 18 | 81 | 01 | 00000001 | 10000001 | 00000001 |  |
| 107 | MOSI | 1A | W | 1A | 19 | E0 | 1Ah (26d) | 00011001 | 11100000 | Recommended on datasheet |
| 107 | MISO | 01 |  | 1A | 01 | 01 | 00000001 | 00000001 | 00000001 |  |
| 106 | MOSI | 1B | W | 1B | 13 | 00 | 1Bh (27d) | 00010011 | 00000000 | XI_trim: 00000 |
| 106 | MISO | 01 |  | 1B | 00 | 01 | 00000001 | 00000000 | 00000001 |  |
| 106 | MOSI | 20 | W | 20 | 48 | 00 | 20h (32d) | 01001000 | 00000000 | PREAMBLE_LEN: 010 (3 bytes); SYNCWORD_LEN: 01 (32 bits → {Reg39[15:0],Reg36[15:0]); TRAILER_LEN: 000 (4 bits); DATA_PACKET_TYPE: 00 (NRZ law data); FEC_TYPE: 00 (No FEC); BRCLK_SEL: 000 (keep low) |
| 106 | MISO | 01 |  | 20 | 01 | 80 | 00000001 | 00000001 | 10000000 |  |
| 105 | MOSI | 21 | W | 21 | 3F | C7 | 21h (33d) | 00111111 | 11000111 | VCO_ON_DELAY_CNT: 3Fh (63 us); TX_PA_OFF_DELAY: 11 (4us + 3us); TX_PA_ON_DELAY: 00111 (7 us) |
| 105 | MISO | 01 |  | 21 | 01 | 80 | 00000001 | 00000001 | 10000000 |  |
| 106 | MOSI | 22 | W | 22 | 20 | 00 | 22h (34d) | 00100000 | 00000000 | TX_CW_DLY: 20h (32 us); TX_SW_ON_DELAY: 000000 (0 us) |
| 106 | MISO | 01 |  | 22 | 01 | 01 | 00000001 | 00000001 | 00000001 |  |
| 106 | MOSI | 23 | W | 23 | 03 | 00 | 23h (35d) | 00000011 | 00000000 | RE-TRANSMIT_TIMES: 0011 (3 times); SCRAMBLE_DATA: 0000000 (no Scramble) |
| 106 | MISO | 80 |  | 23 | 80 | 01 | 10000000 | 10000000 | 00000001 |  |
| 108 | MOSI | 28 | W | 28 | 44 | 01 | 28h (40d) | 01000100 | 00000001 | FIFO_EMPTY_THRESHOLD: 01000 (8d); FIFO_FULL_THRESHOLD: 10000 (16d); SYNCWORD_THRESHOLD: 000001 (0 errors allowed) |
| 108 | MISO | 01 |  | 28 | 01 | 81 | 00000001 | 00000001 | 10000001 |  |
| 107 | MOSI | 29 | W | 29 | B4 | 00 | 29h (41d) | 10110100 | 00000000 | CRC_ON: 1; SCRAMBLE_ON: 0; PACK_LENGTH_EN: 1 (1st byte is payload length); FW_TERM_TX: 1; AUTO_ACK: 0 PKT_FIFO_POLARITY: 1 (FIFO flag Active low); CRC_INITIAL_DATA: 0 |
| 107 | MISO | 01 |  | 29 | 01 | 01 | 00000001 | 00000001 | 00000001 |  |
| 128 | MOSI | 2A | W | 2A | FD | B0 | 2Ah (42d) | 11111101 | 10110000 | SCAN_RSSI_CH_NO: 111111 (63d); Rx_ACK_TIME: B0h (176 us) |
| 128 | MISO | 01 |  | 2A | 00 | 00 | 00000001 | 00000000 | 00000000 |  |
| 117 | MOSI | 2C | W | 2C | 08 | 00 | 2Ch (44d) | 00001000 | 00000000 | DATARATE: 1000 (8d → 125 Kbps) |
| 117 | MISO | 01 |  | 2C | 01 | 01 | 00000001 | 00000001 | 00000001 |  |
| 107 | MOSI | AC | R | 2C | FF | FF | 2Ch (44d) | 11111111 | 11111111 |  |
[... cut ...]
| 1,291 | MOSI | A6 | R | 26 | FF | FF | 26h (38d) | 11111111 | 11111111 | reads SYNC_WORD @ 38 (26h) |
| 1,291 | MISO | 12 |  | 26 | 00 | 00 | 00010010 | 00000000 | 00000000 |  |
| 1,201 | MOSI | A7 | R | 27 | FF | FF | 27h (39d) | 11111111 | 11111111 | reads SYNC_WORD @ 39 (27h) |
| 1,201 | MISO | 1A |  | 27 | 12 | 30 | 00011010 | 00010010 | 00110000 |  |
| 1,510 | MOSI | 87 | R | 07 | FF | FF | 07h (7d) | 11111111 | 11111111 |  |
| 1,510 | MISO | 02 |  | 07 | 00 | 20 | 00000010 | 00000000 | 00100000 |  |

I will keep on investigating on this.

Ciao,
Francesco

pascallanger, 2025-11-14:
The RF chip matches my initial assessment being either a LT8900 or LT8910: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/discussions/1072#discussioncomment-12288115
I've already written a LT89 emulator layer. Need to extract it from the nrf24l01 code and add 125Kbps using the CC2500: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/bf38415420519b616c2a4beb849af74519e661fa/Multiprotocol/NRF24l01_SPI.ino#L247

> the first two bytes of the "RC channels packet" do not change over time and over receivers (I tested with two different, in my case the are always 0x0C 0x80): maybe they could carry information of RC channels 9 and 10 on 8 bit each...

The 0x0C following 0x32 is the packet length=number of bytes to follow. You can see the packet length 0x14 for the channels.

I haven't looked at the dumps but the sync word must be written with the TX address somewhere.

> when switching on the transmitter, after all the RF initialiazion registers, the TX broacasts, using REG 0x07, on RF channel 0x21 the list of the RF channels that will be used for hopping:
>     MOSI	07	01 21
>     MOSI	32	14 8A 74 A4 33 23 55 41 0F 73 19 2D 69 37 05 5F 4B 24 56 42 10
> 23 55 41 0F 73 19 2D 69 37 05 5F 4B are the channels used for frequency hopping and the TX always begins with the second one (0x55)

The TX address is probably 8A 74 A4 33 or 24 56 42 10 . I'm wondering if one of the two is the RX ID. The bind is not described so no idea.

wmlex, 2025-11-24:
@motorello Please tell me if there has been any progress in adding support for HotRC receivers? I have such receivers and a 4-in-1 module, and I can help with testing.

motorello, 2025-11-25:
Yes, I'm kinda progressing, in the sense that I have further analyzed the SPI dumps.
Actually, Pascal was right with the TX address: given the SPI transfer `MOSI 32 14 8A 74 A4 33 23 55 41 0F 73 19 2D 69 37 05 5F 4B 24 56 42 10` it seems that
- **14** is the packet length
- **8A** is the packet type (8A is only sent during first "binding" phase, other values are : 80 when sending RC channels information; B5, B6, B7 when -maybe- TX is asking to receiver information, likely TX RSSI, receiver voltage, telemetry in general... I don't have an educated guessing right now)
- **74 A4 33** are the _last 3 bytes of sync word_: the first one is (seems to be) always 96 and the last one (33 in this case) changes with respect to the model chosen in the transmitter (my CT-8A transmitter has 8 models in memory: model1 has 0x30 as last byte, model2 has 0x31 and so on)
- **23 55 41 0F 73 19 2D 69 37 05 5F 4B** are the RF channels for hopping
- last 4 bytes **24 56 42 10** still undisclosed

After 2.4 seconds, the transmitter reads the RX FIFO, two bytes at a time, until it reads 0x10 0x00. Right after, it starts to send the "RC channels packet" `0C 80 4C 64 22 96 B0 84 25 2C 61 09 4B` (0C is the packet length, 80 is the packet type)
The RC channels packet is sent every 240 ms and it comes back to the same RF channel every 1.44 seconds.

With regard to the emulation implementation in multi-protocol module, I first tried to "abuse" the NRF chip by setting the register NRF24L01_06_RF_SETUP to NRF24L01_BR_RSVD, to try to go at 125kbaud, but of course it didn't work.
So I'm now enhancing the CC2500 driver to allow for more flexible configuration and thus allow for "easy" testing. 
At very first, I am focusing on trying to just receive the HotRC packets from the transmitter, but unfortunately I'm still failing :( I think the trick is to understand the correct sequence between IDLE, CAL, RX... 
In a nutshell, my goal for the moment is to just receive something on a known rf channel, with no hopping: in theory, I'm setting up cc2500 for 125 kbaud and 1MHz channel spacing, wide RF receive bandwidth, tuning on RF channel 0x41, setting in SYNC1 and SYNC0 the first two bytes of the LT89 sync word, disabling the packet len filtering and address filtering, set to RX and poll the status byte. My guess is that at least every 1.44 seconds, when the channel hopping sends on ch 0x41, I should see a packet from the transmitter. 

If I ever succeed in doing something that actually works, I will then write an emulation layer for LT8920, following the example from Pascal.
I wanted to also ask @pascallanger : do you know if any of the general purpose pins of CC2500 (GDO0 and GDO2) are routed to some stm32 pins? That would make the implementation of the protocol easier as I could route some flag signals directly to gpio without polling the chip status byte.

If someone is interested, I can commit the changes on the cc2500 driver so that other people can use them if they like.

Also, I wrote a bash function to help me in transforming the txt output of the Sigrok SPI decoder in something that can be put in spreadsheet for analysis. I'm pasting here, maybe can help:

```
function sigrok_spi_to_csv() {
    sigrok_in=$1
    sed -E '/ bits:/d;
    / data:/d;
    s/ SPI: (M[OI]+S[IO]+) transfers: /\t\1\t/g;
    s/-[0-9]+//g;
    s/ /\t/g;
    /^0\tM[OI]+S[IO]+/d' \
    $sigrok_in | sort -n -k1,1 -k2,2r \
    > ${sigrok_in/.txt/.csv}

    awk -v LA_sample_rate=24000000 'BEGIN {
        /* header */
        LA_sample_period_us=1000000/LA_sample_rate
        printf "R/W\tREGd\tREGh MOSI\tTS (us)\tD-TS (us)\tSTATUS BYTE\tPKT0b\tPKT1b\tTS (samples)\tSIGNAL\tREGh\tPKT0\tPKT1\n"
    }

    /* function to convert decimal to binary */
    function d2b(d, n, b) {
        while(n)
        {
            b = d%2b
            d = int(d/2); n--
        }
        return(b?b:0)
    }

    /* calculate the difference in time (us) from the previous MOSI transfer */
    {
        ts[++i] = $1
        if ( NR > 1 )
        {
            if ( $2 == "MOSI" )
                diff = ($1 - ts[i-2])*LA_sample_period_us
            else
                diff = ""
        }
        else
            diff = 0
    }

    /* evaluate if the SPI command i Read or Write */
    {
        regDecimalRW = strtonum("0x" $3)
        andRW = and(regDecimalRW, 128)
        if ( andRW == 0 )
        {
            RW = "W"
            regDecimal = regDecimalRW
        }
        else
        {
            RW = "R"
            regDecimal = and(regDecimalRW, 127)
        }

    }

    /*  if the transfer is MOSI: save RW flag and register value to be printed also for the following MISO (easy to filter in excel), do not print the status byte */
    /*  if the transfer is MISO: use the RW flag, register value from MOSI and print the status byte in bits */

    {
        if ( $2 == "MOSI" )
        {
            RWPrev = RW
            regDecimalPrev = regDecimal
            statusByteInBit = "-"
        }
        else /* MISO */
        {
            regDecimal = regDecimalPrev
            statusByteInBit = d2b(regDecimal, 8)
            RW = RWPrev
        }
    }

    /* print on file */
    {
        printf "%s\t%s\t%02X\t%.2f\t%s\t%s\t%s\t%s\t%s\n",
            RW, regDecimal, regDecimal, ($1*LA_sample_period_us), diff, statusByteInBit, d2b(strtonum("0x" $4), 8), d2b(strtonum("0x" $5), 8), $0
    }' \
    ${sigrok_in/.txt/.csv} > ${sigrok_in/.txt/-enhanced.csv}

    wc -l $sigrok_in ${sigrok_in/.txt/-enhanced.csv}
}
```
Just use as: `sigrok_spi_to_csv Sigrok_Annotation_Export.txt`

Hope to give more updates soon! (and also hope someone can help me :) )

Doom-Eye, 2026-11-12:
Hi everyone,
first of all, thank you so much for all the great work and effort you put into this project!
I wanted to politely ask if there is any update on the HotRC / Rlaarlo protocol support, as I would love to use my RadioMaster MT12 (4-in-1) with these receivers. Also, please let me know if there is any way I can help out or support the development (e.g., by testing things with my setup).
Thanks in advance!