pascallanger/DIY-Multiprotocol-TX-Module #1083: Using IrangeX IRX4 module in PPM for old TX

gmarconi2, 2025-03-05:
Good morning, to modify old TXs in PPM mode it is increasingly difficult to find FrSky XJT modules with which it was easy to extract the telemetry and display it on a display, using the smartport protocol (the web is full of examples), and use receivers in D8 and D16.
I would like to do the same thing with an IrangeX IRX4 module that is easily found, using it in serial mode with tx equipped with OpenTX or EdgeTX the telemetry works perfectly, because I think that it is the software that resides on the transmitter that takes care of converting the telemetry that comes from the module, (I don't know which pins it uses to do this, since the communication is bidirectional in serial mode).
I would need help for the PPM mode and extract the telemetry using the FrSkyX2 accst v2 protocol from pin 5 of the 5-pin connector with which power and PPM are sent to the module.
What options should I choose in _Config.h (I don't know how it is set, except for the choice of protocols in the IrangeX IRX4.
Probing pin 5 with the oscilloscope, there is a signal, I tried to decode it as FrSkyD 9600bps 8n1, but it doesn't work, what could I do?
The module changes the telemetry based on the chosen protocol:
FrSky D8 telemetry FrSkyD 9600bps 8n1?
FrSkyD16 telemetry FrSkyD 56000bps 8n1?
Help

pascallanger, 2025-03-05:
In PPM mode the moule sends the telemetry as a real module would do. FrskYD is using the 9600 bauds so this is what it sends. A FrskyX2 module sends at 57600 bauds so this is what the module does as simple as that. The FrSkyX2 telemetry frames are exactly matching what an original module would do. So if you find online a decoder it will just work. If you don't find one, as you pointed out the code to decode the frskyx2 telemetry in in OpenTX, EdgeTX, er9x, erskytx and others.

gmarconi2, 2025-03-05:
Ok I try to compile the firmware and send it to the IrangerX IRX4, because I don't know how they compiled it by default.

In *** PPM MODE SETTINGS ***/
What concerns telemetry is:
#define TELEMETRY_FRSKYX_TO_FRSKYD
And associate the protocols to the rotary Switch, the FrSky protocols in option have 40, what does it mean?

The included protocols, instead, are chosen in the section
*** PROTOCOLS TO INCLUDE *** and are valid both in serial mode and in PPM, if I choose for example #define FRSKYX_CC2500_INO, does this also include the sub protocols?

Can I disable the telemetry protocols that I don't use?
Can I disable the RF CHIPS that I don't use?

I'll stop here because there are too many questions.

Thanks in advance Giampiero

pascallanger, 2025-03-05:
If you compile your own firmware with this line defined #define TELEMETRY_FRSKYX_TO_FRSKYD then the module will send everything with the FRSKYD format which means 9600 bauds but you'll only have a few sensors like RSSI and A1,A2.
This is written in the config file: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/fa055e991ce3754dfe7fa9a079ff9b679e092175/Multiprotocol/_Config.h#L433

As you can see there is no selection anywhere for subprotocols in the config file which means that they are all coming with the protocol.

You can disable telemetry protocols that you don't use this is why they are available to define or undef as written in the config file: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/fa055e991ce3754dfe7fa9a079ff9b679e092175/Multiprotocol/_Config.h#L342
Note: If you disable the protocols their associated telemetry is automatically disabled.

For the RF question the answer is also in the config file: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/fa055e991ce3754dfe7fa9a079ff9b679e092175/Multiprotocol/_Config.h#L76

pascallanger, 2025-03-05:
When using PPM, you **MUST** compile the firmware by yourself since a number of parameters have to be adjusted like the PPM endpoints, number of channels, channels order, banks and protocols configuration which is based on your own setup...
The automated builds are mainly for radios using serial and PPM builds are just there as a baseline to verify that the module is working but not more than this.

gmarconi2, 2025-03-06:
Good morning for bank 1 I would like to choose the ACCST V2 protocol in the choice of protocols I only see :
            #define FRSKYX_CC2500_INO nothing like FSRKYX2, I assume it is included.

For the choice with the rotary switch I entered this, is it correct?

/*	5	*/	{PROTO_FRSKYX2,	CH_16		,	0	,	P_HIGH	,	NO_AUTOBIND	,	40	,	0x00000000 },	// option=fine freq tuning
/*	6	*/	{PROTO_FRSKYX2,	EU_16		,	0	,	P_HIGH	,	NO_AUTOBIND	,	40	,	0x00000000 },	// option=fine freq tuning
/*	10 */	{PROTO_MLINK,	  NONE		,	0	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },

romoloman, 2025-03-06:
Hi pascallanger, related to the question on Giampiero is it possible by some config parameter to get telemetry, while using PPM input,  always in FrSky-X protocol even if not using frsky in tx protocol? 
It will be really interesting for old radio conversion to be able to make a simple telemetry monitor without adding the complexity of multitelemetry decoder as in opentx/edgetx.

pascallanger, 2025-03-07:
> Good morning for bank 1 I would like to choose the ACCST V2 protocol in the choice of protocols I only see : #define FRSKYX_CC2500_INO nothing like FSRKYX2, I assume it is included.
> 
> For the choice with the rotary switch I entered this, is it correct?
> 
> / * 1 * / {PROTO_FRSKYX2, CH_16 , 0 , P_HIGH , NO_AUTOBIND , 40 , 0x00000000 }, // option=fine freq tuning / * 2 * / {PROTO_FRSKYX2, EU_16 , 0 , P_HIGH , NO_AUTOBIND , 40 , 0x00000000 }, // option=fine freq tuning / * 3 * / {PROTO_MLINK, NONE , 0 , P_HIGH , NO_AUTOBIND , 0 , 0x00000000 },

It might do what you want. If you use an iRangeX module then probably the freq tuning should be 0 instead of 40.

romoloman, 2025-03-07:
> Nearly all the protocols do output Frsky telemetry for that purpose. The only 2 that don't are DSM and HoTT. [This is what's called HUB TELEMETRY in the config file](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/fa055e991ce3754dfe7fa9a079ff9b679e092175/Multiprotocol/_Config.h#L345)

Thanks for the answer, I did already noticed that but my question was about using the faster frsky sport protocol as output for all the modulations. At the moment for what I do understand it's not possible and i will go on using frsky hub.
I'm making up and I will publish soon on github a telemetry dashboard based on "cheap yellow display".  A touch screen with an esp32 commonly found in several online marketplace with prices ranging from 8 to 14 €...