pascallanger/DIY-Multiprotocol-TX-Module #1086: CADA bricks protocol

pascallanger, 2025-03-09:
Hello,
besides the fact that i am a böoody beginner, i was happy to see that mouldking brick RC boxes are supported .
It would be nice if this will be possible for CADA Bricks RC boxes too ;-)

_Originally posted by @marcrupprath in https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/discussions/1080_

pascallanger, 2025-03-09:
@marcrupprath I've received the CADA set. Quickly before leaving home yesterday, I checked the content. The documentation stated that the receiver brick should be charged before use so I took it with me. When plugging it to a charger (tried a charger and my PC), it flashes red and green. I left it plugged ike this for almost 3 hours and the flashing pattern hasn't changed. Is it expected? The documentation talks about a blue light when charging...

pascallanger, 2025-03-09:
I don't have the small TX with me to test if it works. It will have to wait later this week.

marcrupprath, 2025-03-09:
> [@marcrupprath](https://github.com/marcrupprath) I've received the CADA set. Quickly before leaving home yesterday, I checked the content. The documentation stated that the receiver brick should be [...]

I am not quite shure having found this comment on facebook:https://www.facebook.com/groups/cadabricks/posts/4510688575696227/

It seesm that red /green is ok during charging but i did not know how long it may take,
Checked my boxes : Having red/gren flashlight too.

Even if my boxes may not charged completely ist works fine together with Handsets.
Note about jandset: The green LED is blinking until binding phase is complete.
After that it is only tuned on as long you press a button. That is as it looks on my handsets ;-)

Good luck

marcrupprath, 2025-03-10:
Never mind, due it on your own pace ;-). We are not in a hurry , and we do not have a finishing date for this .

marcrupprath, 2025-04-16:
I would like to ask if there is any progress and what is your assessment? Do you think it is still worth trying to implement CADA?

pascallanger, 2025-04-16:
I need to take the time to rebuild my laptop... It's so broken currently that half of the applications don't work anymore so I can't check anything the radio is sending.

marcrupprath, 2025-04-27:
I forgot to wish you good luck.
Meanwhile i am still busy with 3 additional different Mould King based sets ;-)