pascallanger/DIY-Multiprotocol-TX-Module #1087: Fix XK2 - X4 protocol for XK A300

rdba2k, 2025-03-09:
Hi, Pascal:

I have XK A300 and I verified my code about the gyro-off (senior) mode. The change is in XK2_ccnrf.ino

		//Flags
		packet[5] = GET_FLAG(CH5_SW, 0x01)								//Rate
				  | GET_FLAG(CH6_SW, 0x08)								//Mode
				  | GET_FLAG(CH7_SW, 0x20)								//Hover
				  | GET_FLAG(CH8_SW, 0x40) 								//Light
 				  | GET_FLAG(CH9_SW, 0x10);    // XK A300: Gyro off (Senior mode), only works when CH6=-100%  

The XK A300 stock TX does not send correct code to bring the plane into gyro-off mode even though the manual says so. I test on all flag bits and found out.
You may want to mention that CH7 (Hover) is not used for XK A300 or A280(?) etc, CH9 is used for gyro-off mode, it only works when CH6 is -100%, If CH6 is 100%, CH9 has no effects.

Thanks.

pascallanger, 2025-03-09:
So Iet's make ch6 3 positions.
What are the current 2 modes of ch6? -100%=?, +100%=?

rdba2k, 2025-03-10:
6G: ch6=-100, flag 0x00
3d: ch6=0, flag 0x08 (current code of 2 positions, 3d is at ch6=100)
gyro-off (senior mode): ch6=100, flag 0x10

pascallanger, 2025-03-12:
Could you test the latest code?
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#xk2---99

rdba2k, 2025-03-13:
Bench test of both XK A300 and QF102 are good. I will find time to do the flight test soon.

Thanks.

pascallanger, 2025-03-13:
For the QF102 we could add the telemetry. I can prepare the code but you'll need to put the module in debug mode so that we can see what the telemetry packets look like. The other option is to wait for my order to arrive and I'll look at the telemetry then.

rdba2k, 2025-03-27:
So far XK A300 and QF102 fly well with 4 in 1. I found in XK A300's manual that it also mentions low voltage alert in TX. Do you want to add the telemetry code for XK A300 and I will try to catch the telemetry signal?

Thanks.

rdba2k, 2025-03-27:
This is what I get from XK A300, the airplane's LEDs start flashing when lower than 7.0v. But what I get in the serial monitor does not have any change.

Protocol selected: 99, sub proto 0, rxnum 0, option 0
ID: 26 41 9A CC CC, OFFSET: A0, HOP: 41 45 49 4D
Using CC2500
Proto=XK2, nbr_sub=2, Sub=X4, Opt=2, FS=0, CHMap=0, rfSw=2
RX 9B 26 41 9A 70 36 30 00 31
R:RX_ID=D6
Protocol selected: 99, sub proto 0, rxnum 0, option 0
ID: 26 41 9A CC CC, OFFSET: A0, HOP: 41 45 49 4D
Using CC2500
Proto=XK2, nbr_sub=2, Sub=X4, Opt=2, FS=0, CHMap=0, rfSw=2
RX 9B 26 41 9A 70 36 30 00 31
R:RX_ID=D6

pascallanger, 2025-03-28:
There is no code to receive the telemetry in the current XK2 protocol therefore you won't see anything in debug mode...
I already tried to add telemetry without much success.
On the original set when the TX and RX talks = telemetry is received, the original TX stops hopping from one frequency to another to stay on one fixed channel until may be it doesn't receive telemetry anymore and it starts to hop again...
If I remember I was not getting any telemetry which I assumed might have been based on the fact that I hop from one frequency to another. But later on I had an idea which I haven't had time to look into. The idea is that the flag in packet[6] might not be to tell the RX that I've received its telem packet but instead it's a request from the TX to receive a telemetry packet. Since I've set this flag to 0 in my code, it might explain why I was not receiving anything.
If I have time I'll look into it.

If you really want to help to decode the telemetry from the plane use instead XN297Dump as if I remember the the telem packets are easy to spot.
This is what I've seen:
RX on channel: 69, Time:  3408us P: 66 4F 47 00 00 00 00 00 C8
P[0] = TX_ID[0]
P[1] = TX_ID[1]
P[2] = TX_ID[2]
P[8] = sum P[0..7] + CC
When low batt occurs there must be one value changing somewhere in the packet instead of the 00s in P[4..7].

pascallanger, 2025-03-28:
Just tested, that was it... I had the feeling but after removing all the receiving code and did not feel like redoing everything... So if you set the flag in packet[6] then the plane sends telemetry every 4 packets. My guess is that the plane only listens to 1 RF frequency but I still send on all 4 anyway.
I'll check now what's changing when there is a low batt condition.

pascallanger, 2025-03-29:
Can I close this issue ?

rdba2k, 2025-03-29:
sure.

closed 2025-03-29 by pascallanger