pascallanger/DIY-Multiprotocol-TX-Module #1093: Kyosho EVO 2 Futaba TFHSS

izhelev93, 2025-03-17:
Hello,
TFHSS protocols are quite old now and especially now there is a commonly used rx for Mini-Z by Kyosho - part 82044 using TFHSS protocol. I would like to ask if there is any solution to run it on MT12 MPM or if it could be worked on?

pascallanger, 2025-03-18:
The receiver alone is not helpful... You need both a TX and a RX.

izhelev93, 2025-03-18:
So maybe it will never happen since nobody who owns such helped on that…

closed 2025-03-21 by pascallanger

izhelev93, 2025-03-22:
Hello @pascallanger, does this mean kyosho’s TFHSS is already supported?

pascallanger, 2025-03-22:
No it means that if no one is providing the hardware I can't do anything...

izhelev93, 2025-03-22:
@pascallanger  I have a friend with the whole hardware Futaba 10px tx + same rx as mine. Is there any way I can provide you with data from his hardware or whatever it is needed so that you can do it?

pascallanger, 2025-03-31:
You need to ask him if he is willing to open his TX, solder a logic analyzer to the RF component and do different captures.

pascallanger, 2025-12-05:
A 8ch logic analyzer like this (saleae compatible): https://fr.aliexpress.com/item/1005007112122595.html

tegots123, 2025-12-16:
Just bought the TX and logic analyzer. I imagine i also need a compatible receiver binded to the tx, right ? I Just have the tfhss sr... but will also order a regular tfhss if needed