pascallanger/DIY-Multiprotocol-TX-Module #1097: add telemetry support to SGF22 protocols

rdba2k, 2025-03-29:
Hi, Pascal:

I found out there are also low battery telemetry for SG F22 and KF700 J20. Do you want to send me the modified SGF22 code to collect the telemetry?  It is a little bit hard for me to understand the telemetry code to copy from FX to SGF22 now. Thanks.

pascallanger, 2025-03-29:
I haven't implemented the telemetry since the SG-F22 shows clearly with the inboard LEDs that the battery is low. 
I can add it if you really think its useful...

rdba2k, 2025-03-29:
Let us try it, especially for KF700 J20 which does not have LED.

pascallanger, 2025-03-29:
Ok I'll add the code when I'll have some free time.

pascallanger, 2025-03-29:
Just tried to add the telemetry code but strangely I don't get anything at all from my SGF22... I don't have the original TX anymore so I'm a bit stuck at this point.
Can you do a XN297dump with the original radio and SGF22 plane turned on?
Can you run it in Auto mode?
Can you also scan manually and slowly all the rf frequencies using the XN297dump sub protocol 1M?
Please post both to see if the telemetry is captured and how it looks like.

rdba2k, 2025-03-30:
SGF22 manual says 

4. The remote control will make a "drop" "drop" sound when the aircraft is out of power.

I connected a variable voltage supply to SG F22, its servo starts shaking around 2.9v, I even lower the voltage more, but I did not hear any beep or "drop" sound from the stock TX. Probably there is no telemetry for SGF22. Attached is the dump file. BTW, how to "slowly all the rf frequencies using the XN297dump sub protocol 1M"?  But I can here beep from KF700 J20 stock TX when the voltage is low.

[TX+airplane+VV.txt](https://github.com/user-attachments/files/19523986/TX%2Bairplane%2BVV.txt)

rdba2k, 2025-03-30:
Attached is for KF700 J20 dump

[J20TX+Plane+6v.txt](https://github.com/user-attachments/files/19524056/J20TX%2BPlane%2B6v.txt)

, close to the end the voltage is lower to 6.6v and the stock TX is beeping. I do not think the file shows anything.

pascallanger, 2025-03-30:
Attached files are not interesting, I really need the manual scan....
Select the xn297dump protocol, select the 1M sub protocol then on the field RF frequency which is at 0 by default, increment the value slowly by 1 until you reach the max of 84. You will see the RF channel changing accordingly on the serial terminal.

rdba2k, 2025-03-30:
J20TXNobeep is when the battery voltage is above 7.0v.
J20TXbeep is when the voltage is 7.0v

Nobeep2beep is only monitor on Ch30, it is found this line

RX: 48859us C=30 Enhanced pid=2 S=Y A= C7 95 3C BB A5 P(3)= 00 0B 01

also this line in  J20TXbeep.

It seems only send once. Even I raised the voltage again, the TX beep would not stop.  

[J20TXNobeep.txt](https://github.com/user-attachments/files/19527160/J20TXNobeep.txt)

[J20TXbeep.txt](https://github.com/user-attachments/files/19527169/J20TXbeep.txt)

[nobeep2beep.txt](https://github.com/user-attachments/files/19527172/nobeep2beep.txt)

rdba2k, 2025-03-30:
another collection, "00 0B 01" is found in Ch15.

[auto.txt](https://github.com/user-attachments/files/19527684/auto.txt)
[ch61.txt](https://github.com/user-attachments/files/19527683/ch61.txt)
[ch45.txt](https://github.com/user-attachments/files/19527681/ch45.txt)
[ch30.txt](https://github.com/user-attachments/files/19527680/ch30.txt)
[ch15.txt](https://github.com/user-attachments/files/19527682/ch15.txt)

pascallanger, 2025-03-31:
Does the SGF22 do the same? Since I have it I could try my code...

rdba2k, 2025-04-01:
I do not get any telemetry from SG F22. There is low battery beep for Park 10 F22S, but I do not find any related telemetry signal either.

rdba2k, 2025-04-01:
For J20, I found "00 0B 00" for 8.0v good battery. Apparently ""00 0B 01" for low battery.

Channel=15,0x0F
RX:     0us C=15 Enhanced pid=2 S=Y A= C7 95 3C BB A5 P(12)= 1E 16 00 0B 03 7F 7E 80 84 00 42 10
RX: 48082us C=15 Enhanced pid=2 S=Y A= C7 95 3C BB A5 P(12)= 1E 1A 00 0B 03 7F 7E 80 84 00 42 10
RX: 48083us C=15 Enhanced pid=2 S=Y A= C7 95 3C BB A5 P(12)= 1E 1E 00 0B 03 7F 7E 80 84 00 42 10
RX: 48088us C=15 Enhanced pid=2 S=Y A= C7 95 3C BB A5 P(12)= 1E 22 00 0B 03 7F 7E 80 84 00 42 10
RX: 48785us C=15 Enhanced pid=1 S=Y A= C7 95 3C BB A5 P(3)= 00 0B 00
RX: 47375us C=15 Enhanced pid=2 S=Y A= C7 95 3C BB A5 P(12)= 1E 2A 00 0B 03 7F 7E 80 84 00 42 10
RX: 48086us C=15 Enhanced pid=2 S=Y A= C7 95 3C BB A5 P(12)= 1E 2E 00 0B 03 7F 7E 80 84 00 42 10
RX: 48091us C=15 Enhanced pid=2 S=Y A= C7 95 3C BB A5 P(12)= 1E 32 00 0B 03 7F 7E 80 84 00 42 10
RX: 48081us C=15 Enhanced pid=2 S=Y A= C7 95 3C BB A5 P(12)= 1E 36 00 0B 03 7F 7E 80 84 00 42 10

rdba2k, 2025-04-01:
For J20, telemetry is found only in Ch15 .

[ch15_2.txt](https://github.com/user-attachments/files/19555758/ch15_2.txt)

pascallanger, 2025-04-01:
I've added some code for the telemetry but I haven't tested anything...
Please try first the new code leaving the telemetry flag disabled to see if the protocol still works on the SGF22 and J20.
Then uncomment this line and try again to see if you get some telemetry:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/13024e3816e67567a8295bec711d1ff58c21aa09/Multiprotocol/SGF22_nrf24l01.ino#L52
I'm not quite sure it will work as it is but let's see.

rdba2k, 2025-04-02:
Attached it is for park10 f22s. I only caught "P(3)= 29 B4 00", It seems in the format of rx_tx_addr[2], rx_tx_addr[3],  0x00/0x01(for battery). The telemetry is not constantly sent,  and I do not know which channel it will be sent.

[ch45_f22s.txt](https://github.com/user-attachments/files/19559351/ch45_f22s.txt)

rdba2k, 2025-04-02:
It looks great. ParkTen F22S telemetry works, the low battery voltage threshold is 2.9v (max 4.2v).  KF700 J20  telemetry works too, the low battery voltage threshold is 7.0v (Max 8.4v). I do not see any A1 value change for SG F22. You can verify. And the airplane controls work fine with telemetry code.

pascallanger, 2025-04-02:
I don't have access to my SGF22 anymore for a week or so, I can't test myself.
I've pushed the code with the telemetry enabled by default and updated the documentation if you can give it a last check.
Closing for now as I think we are ok but let me know.

closed 2025-04-02 by pascallanger

rdba2k, 2025-04-03:
The battery low voltage alert works great, but I constantly got "telemetry lost"/"telemetry recovered".  It seems the airplane need be quite close to get the telemetry signal. In edgetx I checked "Disable telemetry audio warning" to deal with this issue.

pascallanger, 2025-04-14:
@rdba2k For the V761, I had to trick the radio since if you miss one or two telemetry packets it was complaining lost/recover. Do you think I should do the same or it's really distance?

rdba2k, 2025-04-15:
We could try to see whether it alleviates the telemetry-lost issue or not. 

Thanks.

pascallanger, 2025-04-16:
Can you test the latest code?
I'm maintaining the telemetry "up" for about 3.5s before declaring it lost to the radio.

rdba2k, 2025-04-18:
The "telemetry loss" situation is improved, but not completely gone, especially when the plane is in aerobatics or flies close to the ground. However, I have received the "low battery" in all situations.

pascallanger, 2025-04-19:
I can increase the wait time...