pascallanger/DIY-Multiprotocol-TX-Module #1099: Kyosho Minium AD PERFEX TX-20 Protocol

StargazerDSM, 2025-03-31:
Is there a way to bind my TX16s to Kyosho Minium AD (4CH) series ? 
I've tried Kyosho2 TK-17 and FHSS , FUTABA and FLYSKY protocols but no luck.
This is the M/C used in the radio 
[image]
Than is on the transmiter board
[image]

Thank you :)

pascallanger, 2025-03-31:
The NRF24L01 shows that this protocol should be doable. FYI, same RF component as the Kyosho2 protocol.
Do you have a logic analyzer to connect to it? You can find one online for a few $/€/... if you are interested.
Otherwise I need the TX and RX shipped to me.

StargazerDSM, 2025-03-31:
I am interested ! I don't have an analyzer but i will buy one.
Is a cheap 25MHz  one ok? 
Do i need to connect it to the receiver also?

StargazerDSM, 2025-04-01:
Front side
[image]
[image]
Back side
[image]

pascallanger, 2025-04-01:
It's really hard to see how things are connected... On the last picture starting on the left, I would skip the 2 first pins, the 3rd one should be ground and then connect the 6 pins to the logic analyzer.
We will see on the first capture if we see what's needed.

KirkTz, 2025-07-23:
Hello, I'm also interested for the kt-20 protocol  ,is that the analyzer ? ,and what software do i need to take the measurements .thnx 
https://vi.aliexpress.com/item/1005006375026372.html

pascallanger, 2025-07-23:
Ah ok, so connect the logic analyzer to it and we'll see what's the difference. They use the exact same RF chip than Kyosho2=KT-17 so it might be a variation of the protocol and not a fully new one.