pascallanger/DIY-Multiprotocol-TX-Module #1103: telemetry missing for QF009 SU35

rdba2k, 2025-04-08:
Hi, Pascal:

I could not find any telemetry signal in TX using current code of 1.3.4.50 for QF009 SU35.

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#MT99XX2---92

The code looks good to me. When the bat. is good, the signal is

P(9)= 91 A7 C0 00 00 00 00 00 A3

when the bat. is low, 

P(9)= 91 A7 C0 80 00 00 00 00 23

Attached is some collection from xn297dump.

Thanks.

[su35.txt](https://github.com/user-attachments/files/19643747/su35.txt)

pascallanger, 2025-04-08:
Currently, the telemetry is only enabled when the Dragon sub protocol is selected: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#sub_protocol-dragon---6
The text file is confusing with the packets length of 4 instead of 9. I'm wondering if that's true or an artefact... To see if it's an artefact, move throttle or any other sticks and see if you still get these 4 bytes payload.
Can you redo dumps using the Auto sub protocol and plane off so I can have a better understanding of the timing?
The timing must be completely different from what it is today but still somewhat/enough working for the RX to operate. But that's bad so you should pay attention to the timing on top of looking at the flags (same was true for the QF012).
The Dragon sub protocol telemetry is also based on an exchange between the TX and the plane. So if they have done the same here we need to look into the details.

rdba2k, 2025-04-08:
[su35_planeoff.txt](https://github.com/user-attachments/files/19653584/su35_planeoff.txt)

rdba2k, 2025-04-08:
[su35_planeoff_1M.txt](https://github.com/user-attachments/files/19653681/su35_planeoff_1M.txt)

rdba2k, 2025-04-08:
When the plane is on, I do not see any signal other than on Ch59. The alert seems at 6.5v.

[su35_planeon_1M_onlyCh59.txt](https://github.com/user-attachments/files/19653855/su35_planeon_1M_onlyCh59.txt)

 
[su35_planeon_1M.txt](https://github.com/user-attachments/files/19653839/su35_planeon_1M.txt)

rdba2k, 2025-04-08:
Another round I only see signal on Ch39.