pascallanger/DIY-Multiprotocol-TX-Module #1106: Telemetry for SG F-22?

stm32h757, 2025-04-11:
Hello,

I successfully bound and flew my SG F-22 (BM-22). All channels are working well with your code—thank you so much for that! It’s a huge help in flying the most meaningful plane I’ve ever owned.

However, the low-battery telemetry isn’t working as expected—it’s different from what’s described in the documentation. You can see it in the video below:  
https://youtu.be/K6nnuakvh2I?si=OVtHvGzLtPZsHwqM

I also found this related issue on GitHub:  
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/1097

It looks like the discussion there has already ended, but I believe it might be worth taking another look.

Is there anything I can do to help?

Although I’m new to the TX16S, I have experience with programming and embedded systems. If you can explain what to check or do, I’ll follow your instructions and share the log files with you.

pascallanger, 2025-04-11:
It looks like the SGF22 only sends telemetry when the battery is low. Please try under this condition.

closed 2025-04-11 by pascallanger

rdba2k, 2025-04-11:
[image]

[image]

Test case:
Normal: 3.9v, A1: 13.2
Low: 2.8v, A1: 0.0

pascallanger, 2025-04-11:
Ok so my guess is that @stm32h757 you are not using the latest test fw...

stm32h757, 2025-04-11:
Yes, I was using 1.3.4.31(latest stable),

I will try test fw soon, Thank you.

stm32h757, 2025-04-12:
I was able to successfully receive the battery information and set up the battery alarm. Thank you!

**[Question]**

I have a VTBird VTOL that uses the FM30 long-range external module, XK A160 and XK X450.

I rarely fly it and have no real need to, but I’m curious—if I use a test build version(1.3.4.50) of the firmware, could it be risky to fly other planes?

Of course, I’ll test everything carefully first.

pascallanger, 2025-04-12:
You can fly with it no problem. On top of it you should now have telemetry for the xk a160 too with this version