pascallanger/DIY-Multiprotocol-TX-Module #1108: Use PPM Mode with old transmitters but use bluetooth telemetry serial connection to configure.

Jonimoose, 2025-04-22:
Like #1083 I picked up an IrangeX module to use with some older TX's to be able to control my newer RC's with my preferred transmitter setup. I've had success with the few vehicles I have and am planning to see about dumping some additional protocols for the vehicles that are not supported when I get a chance. While going through the documentation I saw there is the ability to transmit the telemetry data via a bluetooth serial connection and I plan on picking up the hardware to do this but that made me wonder if there was a way to allow this to be two way so that you could configure the protocol and maybe allow for initiating binding via the serial connection but still using PPM for the channel data.

I expect I would have to make some minor modifications to the board to use a different pin for the serial RX and then obviously adjust the code to watch the serial payload only for protocol changes and possibly other values such as fine tuning or the command to initialize binding but keeping the channel data being sent via ppm?

Relatedly it appears the link to the sample Android app in the bluetooth telemetry pages appears to be broken, do you happen to have another link to this or information on where I could find that information to start tinkering? Though after reading through #1083 building something like that might be a better option to achieve the same goals I have though I again think being able to have some data going back to the 4in1 module would be a nice addition.

scriptforfivem, 2025-05-10:
The documentation in this repository is excellent! Very clear and comprehensive.

fivemhostingserver, 2025-05-10:
Impressive project structure! The architecture really demonstrates solid design principles.

fivemserverhostings, 2025-05-10:
The performance optimizations here are quite clever. Nice engineering work!

fivemeyestore, 2025-05-10:
Interesting solution to this problem! I've been working on similar challenges recently.

fivemtebex, 2025-05-10:
Impressive project structure! The architecture really demonstrates solid design principles.

eyesgodcode, 2025-05-10:
The performance optimizations here are quite clever. Nice engineering work!