pascallanger/DIY-Multiprotocol-TX-Module #1110: Special function 'set failsafe', 'custom' button not visible in module config dialog.

degooglified, 2025-05-06:
Manual says:

"SetFailsafe - Sets the custom failsafe values for the selected module (Internal/External) to the current stick position when activated. For this option to work, the Failsafe mode for the RF module must be set to custom."

I do not find the 'custom' button. How to visualise it?

Tx.

Config: Jumper T18pro, v2.10.6, v1.3.4.31 TAER in both intaernal and external Multi modules.

pascallanger, 2025-05-06:
Which protocol ?

degooglified, 2025-05-06:
DSMX and Radlink.

pascallanger, 2025-05-06:
Dsmx failsafe is done on the RX itself
Radiolink failsafe is done on ch8-16 as indicated in the doc

closed 2025-05-06 by pascallanger

degooglified, 2025-05-06:
Thanks. I do not understand the doc regarding failsafe. Is the failsafe value to be set via sbus in chan 9-16?

degooglified, 2025-05-06:
Pls ignore previous question. Should a fixed FS value be set on 9-16 channels in the transmitter?

degooglified, 2025-05-06:
I found it. Set, special functions to override and set fs fixed values in ch 9-16