pascallanger/DIY-Multiprotocol-TX-Module #1114: Support for Radiolink R12F and R16F receivers

RCBob, 2025-05-28:
Hi Pascal,

Is there any possibility to update the protocol for support of Radiolink Air receivers R12F and R16F, to unlock the respective full 12 and 16 channels capability? As it is, only the first 8 channels are currently available for use.

Both receivers are utilising FHSS 67ch frequency hopping spread spectrum.

Thank you!

pascallanger, 2025-05-28:
I would need a dump of the cc2500 spi bus using a cheap logic analyzer ($10) or send me the TX and RX or participate to the purchase of a set.

gulsennihat, 2025-09-08:
Hello Pascal @pascallanger,
I’m ready to make a donation for Multimodule R12F and R16F support. This matter is very important and urgent for me. Could you please let me know the donation method and amount?
Thank you for your effort and attention.