pascallanger/DIY-Multiprotocol-TX-Module #1120: WPL D12 KEI Truck Protocol

Pencilvestr, 2025-07-24:
Looked through the supported protocols and didn't see anything listed for the WPL D12 Kei Truck. 
Took apart the transmitter. Can't seem to find anything with google matching the IC PN:  2410CTd

[image]

I have a USB logic analyzer, would just need to know where to hook everything up

Thanks!

pascallanger, 2025-07-24:
This is an all in one rf chip so there is nothing to connect to...
If you have an external module, we can try the xn297dump protocol otherwise I need the TX and RX to work on it.

Pencilvestr, 2025-07-24:
Played around with it a little. The Binding sequence for the car is as followings. 
Turn RX / TX on. It bind only after the steering wheel is turned left/right then the car is bound

Here is the my capture with just the TX on. I get nothing until I turn the steering wheel left / Right

'''Protocol selected: 63, sub proto 1, rxnum 0, option 42
XN297 dump, address length=5, bitrate=1M
Proto=XN297DP, nbr_sub=7, Sub=1Mbps , Opt=8, FS=0, CHMap=0, rfSw=2
Channel=42,0x2A
RX:     0us C=42 Bad CRC
RX: 1016305644us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 3D 38 32 80 00 88 00 00 00 00 00 0B
RX: 39480us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 7F 38 32 80 00 88 00 00 00 00 00 4D
RX: 39498us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 7F 38 32 80 00 88 00 00 00 00 00 4D
RX: 39524us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 7F 38 32 80 00 88 00 00 00 00 00 4D
RX: 39489us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 7F 38 32 80 00 88 00 00 00 00 00 4D
RX: 39503us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 72 38 32 80 00 88 00 00 00 00 00 40
RX: 39504us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 09 38 32 80 00 88 00 00 00 00 00 D7
RX: 39540us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39492us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39598us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39448us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39580us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39461us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39544us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39501us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39627us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39454us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39527us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39527us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39523us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 7529618us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 37 38 32 80 00 88 00 00 00 00 00 05
RX: 39506us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 7F 38 32 80 00 88 00 00 00 00 00 4D
RX: 39507us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 7F 38 32 80 00 88 00 00 00 00 00 4D
RX: 39534us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 7F 38 32 80 00 88 00 00 00 00 00 4D
RX: 39501us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 7F 38 32 80 00 88 00 00 00 00 00 4D
RX: 39515us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 7F 38 32 80 00 88 00 00 00 00 00 4D
RX: 39518us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 7D 38 32 80 00 88 00 00 00 00 00 4B
RX: 39510us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 2F 38 32 80 00 88 00 00 00 00 00 FD
RX: 39525us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39660us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39364us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39662us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39378us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39514us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39522us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39712us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39318us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39532us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39527us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39507us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 79019us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39508us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39500us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39517us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39518us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39522us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39504us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39515us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39504us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39523us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39540us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39508us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39519us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
[... cut ...]
RX: 39525us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39528us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39489us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39530us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39517us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX: 39522us C=42 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
'''

pascallanger, 2025-07-24:
Turn on the original TX and RX, make sure you have control. 
Use your radio in debug mode:
Protocol: Multi -> XN297DP
Sub protocol: Auto
Everything unchanged
Let it scan for a while and see if it finds anything.

Pencilvestr, 2025-07-24:
> Turn on the original TX and RX, make sure you have control. Use your radio in debug mode: Protocol: Multi -> XN297DP Sub protocol: Auto Everything unchanged Let it scan for a while and see if it finds anything.

```Detecting XN297 packets.
XN297 dump, address length=5, bitrate=250K
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=1M
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12

Packet detected: bitrate=Using NRF
1M C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
--------------------------------
Identifying all RF channels in use.
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12
RX on channel: 12, Time:     0us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 39525us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 39554us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 39552us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 39545us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 39549us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 79092us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 39531us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 39536us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 39544us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 39553us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
Trying RF channel: ,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29
RX on channel: 29, Time:     0us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39550us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39546us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39541us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39545us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39527us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39580us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39503us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39533us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39535us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39529us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39537us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39533us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
Trying RF channel: ,30,31,32,33,34,35,36,37,38,39
RX on channel: 39, Time:     0us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 39, Time: 237170us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
Trying RF channel: ,40,41,42
RX on channel: 42, Time:     0us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39521us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39515us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39598us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39465us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39511us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39521us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39519us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39531us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39547us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39526us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39525us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
Trying RF channel: ,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58
RX on channel: 58, Time:     0us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 58, Time: 158071us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
Trying RF channel: ,59,60,61
RX on channel: 61, Time:     0us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39521us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39526us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39539us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39520us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39510us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39530us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39524us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39525us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39527us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39519us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39527us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
[... 21 lines cut ...]
Time between CH:29 and CH:42
Time: 19767us
Time: 19716us
Time: 19770us
Time: 19772us
Time: 19791us
Time: 19776us
Time: 19778us
Time: 19762us
Time: 19776us
Time: 19787us
Time: 19764us
Time between CH:29 and CH:61
Time: 29648us

Channel order:
29:     0us
12:  9674us
42: 19716us
61: 29648us
29: 39308us

--------------------------------
Identifying Sticks and features.
P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E```

Pencilvestr, 2025-07-24:
**TX only. No controls touched**
```XN297 dump, address length=5, bitrate=250K
Proto=XN297DP, nbr_sub=7, Sub=Auto  , Opt=8, FS=0, CHMap=0, rfSw=2
------------------------
Detecting XN297 packets.
XN297 dump, address length=5, bitrate=250K
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=1M
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23

Packet detected: bitrate=Using NRF
1M C=23 S=Y A= 69 A5 37 4D 8B P(16)= 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
--------------------------------
Identifying all RF channels in use.
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23
RX on channel: 23, Time:     0us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 23, Time: 39511us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 23, Time: 39503us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 23, Time: 39515us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 23, Time: 39503us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 23, Time: 39505us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 23, Time: 39500us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 23, Time: 39525us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 23, Time: 39518us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 23, Time: 39510us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 23, Time: 39532us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 23, Time: 39502us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
Trying RF channel: ,24,25,26,27,28,29,30,31,32,33,34,35,36,37
RX on channel: 37, Time:     0us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 37, Time: 39515us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 37, Time: 39514us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 37, Time: 39691us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 37, Time: 39355us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 37, Time: 39508us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 37, Time: 39501us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 37, Time: 39500us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 37, Time: 39506us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 37, Time: 39514us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 37, Time: 39510us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 37, Time: 39509us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
Trying RF channel: ,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54
RX on channel: 54, Time:     0us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 54, Time: 78784us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 54, Time: 39487us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 54, Time: 39486us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 54, Time: 39507us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 54, Time: 39469us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 54, Time: 39484us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 54, Time: 39501us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 54, Time: 39479us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 54, Time: 39490us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 54, Time: 39478us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
Trying RF channel: ,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70
RX on channel: 70, Time:     0us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 70, Time: 39510us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 70, Time: 39519us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 70, Time: 39516us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 70, Time: 39510us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 70, Time: 39521us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 70, Time: 39520us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 70, Time: 39509us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 70, Time: 39538us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 70, Time: 39516us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 70, Time: 39517us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
RX on channel: 70, Time: 39527us P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33
Trying RF channel: ,71,72,73,74,75,76,77,78,79,80,81,82,83,84

4 RF channels identified: 23[12] 37[12] 54[11] 70[12]
Keeping only RF channels with more than 8 packets: 23* 37 54 70
--------------------------------
Identifying RF channels order.
Time between CH:23 and CH:23
Time: 39471us
Time: 39500us
Time: 39508us
Time: 39509us
Time: 39496us
Time between CH:23 and CH:37
Time: 18297us
Time:  9863us
Time: 10081us
Time:  9877us
Time: 128353us
Time: 10070us
Time:  9871us
Time between CH:23 and CH:54
Time: 29633us
Time: 29641us
Time: 29636us
Time: 29645us
Time: 29631us
Time: 29633us
Time between CH:23 and CH:70
Time: 11424us
Time: 19744us
Time: 19571us
Time: 19758us
Time: 19760us
Time: 19752us

Channel order:
23:     0us
37:  9863us
70: 11424us
54: 29631us

--------------------------------
Identifying Sticks and features.
P: 94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33```

Pencilvestr, 2025-07-24:
**TX only. Steering wheel moved like trying to "bind"**

```Detecting XN297 packets.
XN297 dump, address length=5, bitrate=250K
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=1M
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12

Packet detected: bitrate=Using NRF
1M C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
--------------------------------
Identifying all RF channels in use.
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12
RX on channel: 12, Time:     0us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 39484us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 39495us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 39503us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 39502us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 39499us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 39509us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 39509us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 39486us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 39504us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 39507us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 12, Time: 39515us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
Trying RF channel: ,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29
RX on channel: 29, Time:     0us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39475us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39488us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39475us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39479us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39478us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39481us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39484us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39494us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39488us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39488us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 29, Time: 39501us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
Trying RF channel: ,30,31,32,33,34,35,36,37,38,39,40,41,42
RX on channel: 42, Time:     0us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39519us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39512us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39511us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39523us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39527us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39526us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39526us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39504us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39513us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39521us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 42, Time: 39518us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
Trying RF channel: ,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61
RX on channel: 61, Time:     0us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39523us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39532us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39531us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39534us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39524us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39530us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39523us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39544us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39536us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39527us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39526us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
RX on channel: 61, Time: 39507us P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E
Trying RF channel: ,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84

4 RF channels identified: 12[12] 29[12] 42[12] 61[13]
Keeping only RF channels with more than 8 packets: 12 29 42 61*
--------------------------------
Identifying RF channels order.
Time between CH:61 and CH:12
Time: 19764us
Time: 19769us
Time: 19772us
Time: 19770us
Time: 19773us
Time: 19754us
Time: 19760us
Time: 19756us
Time: 19751us
Time: 19746us
Time between CH:61 and CH:29
Time: 32321us
Time:  9868us
Time between CH:61 and CH:42
Time: 29640us
Time: 29641us
Time: 29679us
Time: 29652us
Time: 29651us
Time: 29638us
Time: 29643us
Time: 29684us
Time: 29651us
Time: 29645us
Time between CH:61 and CH:61
Time: 39487us
Time: 39551us

Channel order:
61:     0us
29:  9868us
42: 29638us
61: 39487us

--------------------------------
Identifying Sticks and features.
P: 94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E```

pascallanger, 2025-07-24:
In normal and bound mode, when you are at the step: Identifying Sticks and features
Move slowly the steering mid/left/mid/right/mid and share that specific part.
Do the same for the throttle. 
Do the same for any other channels and features.

Pencilvestr, 2025-07-24:
Steering trim full sweep

``RX: 39564us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 22 32 80 00 88 00 00 00 00 00 38
RX: 39566us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 22 32 80 00 88 00 00 00 00 00 38
RX: 39563us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 22 32 80 00 88 00 00 00 00 00 38
RX: 39570us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 22 32 80 00 88 00 00 00 00 00 38
RX: 79132us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 22 32 80 00 88 00 00 00 00 00 38
RX: 79134us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 22 32 80 00 88 00 00 00 00 00 38
RX: 39567us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 22 32 80 00 88 00 00 00 00 00 38
RX: 79128us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 22 32 80 00 88 00 00 00 00 00 38
RX: 39569us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 22 32 80 00 88 00 00 00 00 00 38
RX: 39579us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 22 32 80 00 88 00 00 00 00 00 38
RX: 39556us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 22 32 80 00 88 00 00 00 00 00 38
RX: 39572us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 22 32 80 00 88 00 00 00 00 00 38
RX: 39578us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 22 32 80 00 88 00 00 00 00 00 38
RX: 39575us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 22 32 80 00 88 00 00 00 00 00 38
RX: 39579us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 22 32 80 00 88 00 00 00 00 00 38
RX: 79138us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 23 32 80 00 88 00 00 00 00 00 39
RX: 39576us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 24 32 91 00 88 00 00 00 00 00 4B
RX: 39579us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 25 32 91 00 88 00 00 00 00 00 4C
RX: 39642us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 26 32 91 00 88 00 00 00 00 00 4D
RX: 39614us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 28 32 91 00 88 00 00 00 00 00 4F
RX: 39461us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 29 32 95 00 88 00 00 00 00 00 54
RX: 39593us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 2A 32 AB 00 88 00 00 00 00 00 6B
RX: 39576us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 2C 32 AB 00 88 00 00 00 00 00 6D
RX: 39584us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 2E 32 AB 00 88 00 00 00 00 00 6F
RX: 39551us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 32 32 BD 00 88 00 00 00 00 00 85
RX: 39555us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 34 32 BD 00 88 00 00 00 00 00 87
RX: 39543us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 37 32 BD 00 88 00 00 00 00 00 8A
RX: 39793us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 3A 32 C6 00 88 00 00 00 00 00 96
RX: 39334us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 3C 32 C6 00 88 00 00 00 00 00 98
RX: 39557us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 3F 32 CA 00 88 00 00 00 00 00 9F
RX: 39567us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX: 39556us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 44 32 CA 00 88 00 00 00 00 00 A4
RX: 39574us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 47 32 CA 00 88 00 00 00 00 00 A7
RX: 39570us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 4B 32 CA 00 88 00 00 00 00 00 AB
RX: 39586us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 4E 32 CA 00 88 00 00 00 00 00 AE
RX: 39578us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 52 32 E9 00 88 00 00 00 00 00 D1
RX: 39547us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 57 32 CE 00 88 00 00 00 00 00 BB
RX: 39550us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5B 32 DB 00 88 00 00 00 00 00 CC
RX: 39561us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 E9 00 88 00 00 00 00 00 DD
RX: 39563us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 E9 00 88 00 00 00 00 00 DD
RX: 39543us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 E9 00 88 00 00 00 00 00 DD
RX: 39558us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 E9 00 88 00 00 00 00 00 DD
RX: 39552us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 E9 00 88 00 00 00 00 00 DD
RX: 39533us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 E9 00 88 00 00 00 00 00 DD
RX: 39558us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 E9 00 88 00 00 00 00 00 DD
RX: 39637us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 E9 00 88 00 00 00 00 00 DD
RX: 39722us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 E9 00 88 00 00 00 00 00 DD
RX: 39360us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 E9 00 88 00 00 00 00 00 DD
RX: 39568us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 CE 00 88 00 00 00 00 00 C2
RX: 39550us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 CE 00 88 00 00 00 00 00 C2
RX: 39588us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 E9 00 88 00 00 00 00 00 DD
RX: 39546us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 E9 00 88 00 00 00 00 00 DD
RX: 39562us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 E9 00 88 00 00 00 00 00 DD
RX: 39563us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 E9 00 88 00 00 00 00 00 DD
RX: 39804us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 E9 00 88 00 00 00 00 00 DD
RX: 39530us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 E9 00 88 00 00 00 00 00 DD
RX: 39352us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 E9 00 88 00 00 00 00 00 DD
RX: 39531us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 E9 00 88 00 00 00 00 00 DD
RX: 39563us C=12 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 5E 32 E9 00 88 00 00 00 00 00 DD
```

No throttle trim

pascallanger, 2025-07-26:
Bind packet
-----------
XN297 1Mb Scrambled
Bind address: 69 A5 37 4D 8B
RF channels: 23, 37, 70, 54
Timing: 9875µs
Payload 16 bytes
94 16 CC 96 2A A9 32 B4 00 08 00 00 00 00 00 33

P[0] = 94 ??
P[1] = 16 ??
P[2] = CC ??
P[3..7] = Normal address
P[8] = 00 ??
P[9] = 08 ?? not bound?, Throttle and Steering rate low, Headlights on
P[10..14] = 00 ??
P[15] = sum(P[0..14])+66 why 66...

Normal packet
-----------
XN297 1Mb Scrambled
Normal address: 96 2A A9 32 B4
RF channels: 12=0C, 42=2A, 61=3D, 29=1D
Timing: 9875µs
Payload 16 bytes
94 16 CC 80 80 38 32 80 00 88 00 00 00 00 00 4E

P[0] = 94 ??
P[1] = 16 ??
P[2] = CC ??
P[3] = Throttle, not enough data on dumps... Same coding as Steering?
P[4] = Steering, not enough data on dumps, looks like one side goes from 7F to 00 and the other 80 to FF which would be s8b
P[5] = Steering trim 22..5E, mid gives 40 not 38... Was the trim centered on the other dumps with value 38?
P[6] = 32 ??
P[7] = 80 ?? Additional channel? It moves at the same time as the trim but my guess is that it is an unconnected channel.
P[8] = 00 ??
P[9] = 80 ?? bound?, Throttle and Steering rate low, Headlights off
      |02 -> Steering rate high
      |04 -> Throttle rate high
      |08 -> Headlights on
P[10..14] = 00 ??
P[15] = sum(P[0..14])+66 why 66...

Bottom line: To be able to code this protocol, you need to move far more slowly Throttle and Steering using the Auto sub protocol at the step "Identifying sticks and features" which removes all the duplicates and shows direclty what's changing.
What should be the protocol name?
Any other models using the same TX?

Pencilvestr, 2025-07-26:
It looks like all the WPL models use the same transmitter.
https://www.wpl-rc.com
This model being the "D12"

There is no detent to center the steering trim, I have now adjusted it to where I believe center is

**Header**
```
Protocol selected: 63, sub proto 3, rxnum 0, option 0
XN297 dump, address length=5, bitrate=250K
Proto=XN297DP, nbr_sub=7, Sub=Auto  , Opt=8, FS=0, CHMap=0, rfSw=2
------------------------
Detecting XN297 packets.
XN297 dump, address length=5, bitrate=250K
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84
XN297 dump, address length=5, bitrate=1M
Trying RF channel: 0,1,2,3,4,5,6,7,8,9

Packet detected: bitrate=Using NRF
1M C=9 S=Y A= 96 2A A9 32 B4 P(16)= 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
--------------------------------
Identifying all RF channels in use.
Trying RF channel: 0,1,2,3,4,5,6,7,8,9
RX on channel: 9, Time:     0us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 9, Time: 79100us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 9, Time: 39551us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 9, Time: 39527us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 9, Time: 39529us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 9, Time: 39538us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
Trying RF channel: ,10,11,12
RX on channel: 12, Time:     0us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 12, Time: 39568us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 12, Time: 39538us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 12, Time: 39547us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 12, Time: 39543us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 12, Time: 39570us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 12, Time: 39555us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 12, Time: 39565us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 12, Time: 39556us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 12, Time: 39539us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 12, Time: 39566us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 12, Time: 39536us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 12, Time: 39559us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
Trying RF channel: ,13,14,15,16,17,18,19,20,21,22,23,24,25,26
RX on channel: 26, Time:     0us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 26, Time: 79110us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 26, Time: 158200us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 26, Time: 118641us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 26, Time: 39570us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
Trying RF channel: ,27,28,29
RX on channel: 29, Time:     0us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 29, Time: 39551us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 29, Time: 39559us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 29, Time: 39522us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 29, Time: 39541us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 29, Time: 39546us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 29, Time: 39542us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 29, Time: 39527us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 29, Time: 39545us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 29, Time: 39553us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 29, Time: 39748us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 29, Time: 39323us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
Trying RF channel: ,30,31,32,33,34,35,36,37,38,39
RX on channel: 39, Time:     0us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 39, Time: 39560us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
Trying RF channel: ,40,41,42
RX on channel: 42, Time:     0us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 42, Time: 39454us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 42, Time: 39544us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
RX on channel: 42, Time: 39563us P: 94 16 CC 80 80 40 32 CA 00 88 00 00 00 00 00 A0
[... 661 lines cut ...]
13:35:52.978 -> P: 94 16 CC 80 80 22 32 91 00 8A 00 00 00 00 00 4B
13:35:53.117 -> P: 94 16 CC 80 80 22 32 95 00 8A 00 00 00 00 00 4F
13:35:53.162 -> P: 94 16 CC 80 80 22 32 91 00 8A 00 00 00 00 00 4B
13:35:53.487 -> P: 94 16 CC 80 80 22 32 95 00 8A 00 00 00 00 00 4F
13:35:53.535 -> P: 94 16 CC 80 80 22 32 91 00 8A 00 00 00 00 00 4B
13:35:53.582 -> P: 94 16 CC 80 80 22 32 80 00 8A 00 00 00 00 00 3A
13:35:53.627 -> P: 94 16 CC 80 80 22 32 95 00 8A 00 00 00 00 00 4F
13:35:53.674 -> P: 94 16 CC 80 80 22 32 91 00 8A 00 00 00 00 00 4B
13:35:54.001 -> P: 94 16 CC 80 80 22 32 95 00 8A 00 00 00 00 00 4F
13:35:54.047 -> P: 94 16 CC 80 80 22 32 91 00 8A 00 00 00 00 00 4B
[... cut ...]
13:35:54.465 -> P: 94 16 CC 80 80 24 32 95 00 8A 00 00 00 00 00 51
13:35:54.513 -> P: 94 16 CC 80 80 25 32 91 00 8A 00 00 00 00 00 4E
13:35:54.559 -> P: 94 16 CC 80 80 26 32 91 00 8A 00 00 00 00 00 4F
13:35:54.559 -> P: 94 16 CC 80 80 27 32 91 00 8A 00 00 00 00 00 50
13:35:54.606 -> P: 94 16 CC 80 80 28 32 95 00 8A 00 00 00 00 00 55
13:35:54.652 -> P: 94 16 CC 80 80 29 32 95 00 8A 00 00 00 00 00 56
13:35:54.699 -> P: 94 16 CC 80 80 2A 32 AB 00 8A 00 00 00 00 00 6D
```

Hopefully that data is a little better! 
Thanks!

pascallanger, 2025-07-26:
Thanks, I'll check the dumps and implement the WPL protocol.
From what I can see they have 3 different radios RES-V3, "Basic" and XX-162R: https://www.wpl-rc.com/collections/electronics
From this page, I guess you have the "Basic" one compatible with these models D12, D12KM, D22, D32, D42, D14 : https://www.wpl-rc.com/products/d12-electronics-pack?_pos=2&_sid=d06c4aa09&_ss=r

Pencilvestr, 2025-07-26:
That is correct

pascallanger, 2025-07-27:
Please try the test version 1.3.4.51 : https://downloads.multi-module.org/latest-test/
I haven't checked anything so... and it's using your original TX ID for now.
Doc: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#wpl---107

Pencilvestr, 2025-07-27:
Thanks for all the hard work!

Been messing around with it for the past 20-30 minutes. Doesn't want to bind. 

On the original TX/RX with both turned on, it does not actually bind until you move steering

pascallanger, 2025-07-27:
Ok, let me check the code.

pascallanger, 2025-07-27:
Try again the last test version

Pencilvestr, 2025-07-27:
No luck

Pencilvestr, 2025-07-27:
Bound!
Throttle and steering work, will take me a bit to go thru other functions.

As usual, fantastic work!

Pencilvestr, 2025-07-27:
All other functions work. 

I'm willing if you are

pascallanger, 2025-07-27:
Can you compile the code by yourself?

pascallanger, 2025-07-27:
Can you try to comment this line https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/5ade6bc312e3ae7f5855bd6867613b7f23ba9064/Multiprotocol/WPL_nrf24l01.ino#L21C1-L21C30 by adding "//" in front of it and see if you still can bind?

pascallanger, 2025-07-27:
Ok, try with the new code I've pushed.
If your receiver number is set to 0 then you should find these RF frequencies: 12, 29, 42, 61.
Now set the RX number to 1 and see if you can find the 4 RF frequencies.

Pencilvestr, 2025-07-27:
RX 0...It bound on the following: 12,13, 29, 30, 42, 43, 61, 62
RX 1...Nothing

pascallanger, 2025-07-27:
Ok on this line: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/b88f4d40f46cbf03eb6c71e9288fac5037333279/Multiprotocol/WPL_nrf24l01.ino#L96
Try to change [0] by a [1] and retry with RX number 1. If you could try [2], [3] and [4]

pascallanger, 2025-07-27:
Put back the RX number to 0, and try changing the value of: 
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/b88f4d40f46cbf03eb6c71e9288fac5037333279/Multiprotocol/WPL_nrf24l01.ino#L42
Increment the value by 1 for example and see if you can still bind.
Same for packet [1] and [2]

Pencilvestr, 2025-07-27:
> Just to be sure, you've turned off/on the truck and done a bind each time before trying the frequencies.

Correct

So it would look like this?

```
	packet[0] = 0x94;											//??
	packet[1] = 0x16;											//??
	packet[2] = 0xCC;											//??

Then

	packet[1] = 0x94;											//??
	packet[1] = 0x16;											//??
	packet[2] = 0xCC;											//??

	packet[0] = 0x94;											//??
	packet[2] = 0x16;											//??
	packet[2] = 0xCC;											//??

	packet[0] = 0x94;											//??
	packet[1] = 0x16;											//??
	packet[3] = 0xCC;											//??
```

pascallanger, 2025-07-27:
No, instead of 94 try 95, instead of 16 try 17, instead of CC try CD

pascallanger, 2025-07-27:
At this stage my guess is that the first 3 values are somehow linked to the ID.
I'm not sure what to try next.  I'll let you know.

Pencilvestr, 2025-07-27:
No worries. Again thanks for all the work you do!
I'll make a post on RCgroups and see if anyone else has a WPL and can do a dump for you if that's helpful

pascallanger, 2025-07-27:
That would definitely help to have more examples