pascallanger/DIY-Multiprotocol-TX-Module #1121: Failsafe Issue with TX16S and XY7000 Receiver(Hisky Std Protocol)

stm32h757, 2025-07-25:
- EdgeTX Version : v2.10.6 "Centurion"

- Multimodule version : V1.3.4.50(Test build version, uploaded to the module on 2025.04.12)

- Receiver : XY7000(Quite old... I bought this in Japan during my travels.)
   
[image]

- Original Transmitter : Kyosho H-6K(Quite old... I bought this in Japan during my travels.)

[image]

- Using Protocol with TX16S :
  MULTI - Hisky - Std
  I tried Futaba FHSS and Kyosho FHSS, but neither would bind. The receiver has the FHSS logo.

- Problem Description :

I successfully bound the TX16S to the XY7000, and it works well. However, I encountered one issue: **Failsafe**.

I tried all the failsafe options, but they didn't work. When I turn off the transmitter, the motor (throttle channel) sticks at some value (possibly 1500?).

- What I tried :

I tried setting the failsafe with the original transmitter, but there was no failsafe option available. The failsafe worked correctly with the original transmitter; when I turned it off, the motor (throttle channel) stopped spinning.

pascallanger, 2025-07-25:
Failsafe is not supported for the Std sub protocol, only the HK310 sub protocol as indicated in the doc.
I have no idea how failsafe packets could be sent to the receiver. 10 bytes are sent and nearly every single bit is used (nearly because the values are between 0 and 1000, so one could use the value of 1023 on one or ore channels to set the failsafe for the other channels). I think it is unlikely that the failsafe comes from the TX. Have you tried binding or pressing the bind button (once or twice or more) on the RX with the sticks in the desired failsafe position? That's how most of the old RXs are saving the failsafe positions.

stm32h757, 2025-07-26:
Hello.

Very Thank you for your immediate reply.

(1) My Solution :

Calibrate the ESC with the TX16S, using only the throttle channel form -75% to 100%(Not form -100% to +100%)
It worked, when I turn off transmitter the motor now stop spinning.

Also, this method allow me to utilize the full range of throttle channel(100%). other method such as calibrating with "H-6K", make failsafe function to work correctly, but did not allow full use of the throttle range.

(2) Detailed Explanation :

      (a) As you suggested, I tried every possible way to press the bind button — multiple times, 
           long press, and under various conditions — but none of those methods allowed me to successfully set the failsafe function.

          Each time I bound a receiver, I also calibrated the ESC (FlyFun 12A) accordingly to match the transmitter.
          There was only one consistent pattern:
          When binding with the H-6K transmitter, the failsafe worked properly, the motor stopped as expected.
          But when binding with the TX16S, the servo failsafe behaved correctly, yet the ESC kept spinning the motor at its minimum throttle value.

          It seems that for this receiver, the failsafe behavior is fixed, servos go to neutral, and the throttle channel goes to its minimum value, and this cannot be changed.

          If needed, I can also share my Mixes or other TX16S settings, but I don’t believe there’s anything wrong there.
So far, I’ve successfully configured around 8 aircraft, so I’m fairly confident in the setup.

          I’ve uploaded two videos to show the difference in failsafe behavior between the two setups:

* [https://youtu.be/BImo6plhT3E?si=RaieGIb87myLVEmM](https://youtu.be/BImo6plhT3E?si=RaieGIb87myLVEmM)
* [https://youtu.be/H3kCaptijSY?si=3-tss9VndHQsBaEI](https://youtu.be/H3kCaptijSY?si=3-tss9VndHQsBaEI)

      (b) Also, since the model names "XY3000" and "XY7000" look similar, I wonder if their firmware or internal codebase is also similar.

pascallanger, 2025-07-26:
I don't see how can I help further.
I'm not sending any failsafe packets for the protocol Hisky/Std and I don't know how to send them.
If you need further investigation, I would need to see what the "H-6K" is doing differently. This can be done by doing a SPI dump using a logic analyzer or sending me the radio.
But if you are ok with your setup then please close this issue.