pascallanger/DIY-Multiprotocol-TX-Module #1123: Missing Protocols

KrustyFPV, 2025-08-06:
Hi there i think it is my fault but i cant find out why i dont get some Protocols on my radiomaster MT12 i downloaded
mm-stm-serial-aetr-sfc-v1.3.4.31
New Radiolink Protocol is now there but can not find WPL (107)

can anyone help?

KrustyFPV, 2025-08-07:
i can send a truck(b36) with radio to Pascal if it helps we already did this for building the Mould King Protocol

pascallanger, 2025-08-08:
Hey, to add your original TX ID, I only need the TX to be sent which is the cheapest option even more if you decide that you don't need it anymore so no shipping back.
Having additionally the truck or only the receiver board (to simplify the shipment) it you can extract it, might enable me to find more IDs which is not guaranteed... That means shipping cost both ways and you won't be able to use your truck during that period.
I don't remember how much they are selling the kit radio + receiver but that might be another option to get it delivered directly at my address. 
I'm open to all 3 options. Feel free to contact me by email as you've done in the past.

lschunn, 2025-08-11:
> It's only in the beta builds at the moment https://downloads.multi-module.org/latest-test
> 
> Also if you have the ability you should check out the thread and see if you have the ability to help us get more TX ID's because at the moment we only have mine

how can we check the latest-test build included what kind of changes?

pascallanger, 2025-08-11:
Check the changes since the last release directly on Github:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/compare/v1.3.4.31...master